firmadyne/firmadyne #204: Error: Cannot open file --run-as=root

h0pe-ay, 2022-12-20:
sudo python3 ./sources/extractor/extractor.py -b dlink -sql 127.0.0.1 -np -nk "../firmware/DIR-815_FIRMWARE_1.01.ZIP" images

>> Database Image ID: 9050

/home/pwn/FirmAFL/firmware/DIR-815_FIRMWARE_1.01.ZIP
>> MD5: 1e941413c6489000c7ee6269e98e0a3f
>> Tag: 9050
>> Temp: /tmp/tmp1r3o5jt_
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True

General Error: Cannot open file --run-as=root (CWD: /tmp/tmp1r3o5jt_) : [Errno 2] No such file or directory: '--run-as=root'

General Error: Cannot open file --preserve-symlinks (CWD: /tmp/tmp1r3o5jt_) : [Errno 2] No such file or directory: '--preserve-symlinks'

Traceback (most recent call last):
  File "./sources/extractor/extractor.py", line 427, in extract
  File "./sources/extractor/extractor.py", line 474, in _check_archive
  File "./sources/extractor/extractor.py", line 653, in _check_recursive
  File "/usr/local/lib/python3.7/site-packages/binwalk/__init__.py", line 10, in scan
    objs = m.execute()
  File "/usr/local/lib/python3.7/site-packages/binwalk/core/module.py", line 783, in execute
    obj = self.run(module)
  File "/usr/local/lib/python3.7/site-packages/binwalk/core/module.py", line 802, in run
    obj = self.load(module, kwargs)
  File "/usr/local/lib/python3.7/site-packages/binwalk/core/module.py", line 833, in load
    argv.update(self.dependencies(module, argv['enabled']))
  File "/usr/local/lib/python3.7/site-packages/binwalk/core/module.py", line 865, in dependencies
    raise ModuleException("Failed to load " + dependency.name + " module")
binwalk.core.exceptions.ModuleException: Failed to load General module
>> Cleaning up /tmp/tmp1r3o5jt_...

closed 2022-12-20 by h0pe-ay

1998-zhang, 2023-04-29:
How to solve this problem

MatthewTingum, 2023-06-30:
I think this is `binwalk` issue. IIRC the pypi package is not maintained and is broken in some way. Install `binwalk` from source and it might clear up this issue.

I switched to using Python2.7 from Python3 and in the course of doing so, I had to install binwalk from source. It could have been something else that fixed it for me.

MatthewTingum, 2023-06-30:
To follow up, the following got me going on `python3`:
- `apt remove python2.7`
- `apt remove binwalk`
- `apt install python-is-python3`
- Install `binwalk` from source for Python 3

h0pe-ay, 2023-06-30:
> To follow up, the following got me going on `python3`:
> 
> * `apt remove python2.7`
> * `apt remove binwalk`
> * `apt install python-is-python3`
> * Install `binwalk` from source for Python 3

The method you said is very effective, thank you

Kidder1, 2024-01-18:
i did this ,but it doesn't work how to fix this ?who can help me ?
[image]

h0pe-ay, 2024-03-02:
> i did this ,but it doesn't work how to fix this ?who can help me ? the error is the same as before one person

You should install python3 as well as the latest binwalk