firmadyne/firmadyne #207: inferNetwork.sh UnicodeDecodeError

iAmG-r00t, 2023-01-20:
I am trying to emulate this router [image](http://www.netis-systems.com/Suppory/downloads/dd/1/img/75) but I am getting this error while trying to run the ***inferNetwork.sh*** script. Tried to fix it but I can't see the `codecs.decode(var, 'utf-8')` line so am really confused now.

[image]

iAmG-r00t, 2023-01-20:
So I can see you can get this error while also trying to open a file [source](https://stackoverflow.com/questions/16528468/while-reading-file-on-python-i-got-a-unicodedecodeerror-what-can-i-do-to-resol)

iAmG-r00t, 2023-01-20:
Okay this `data = open(infile, encoding='iso-8859-15').read()` fixes it. I think we should place a try ... except using various types of encoding types

iAmG-r00t, 2023-01-20:
But still unable to emulate the image because it does not have an `/etc/init.d/rcS` or `/etc/rcS.sh` file

hackinaggie, 2023-08-15:
Changing the encoding worked for me as well. Easy little workaround for when the firmware's init scripts give bad output.

commit 748fdec9 (2023-10-31): fix #207 (#216)

Fix the makeNetwork.sh error with Unicode characters when parsing the qemu log file
closed 2023-10-31 by ddcc