firmadyne/firmadyne #217: No images when running the extractor on the sample firmware

myasn1k, 2023-11-07:
Hi, sorry bothering you. 

I'm testing the environment following the "Usage" section of the README but when i run the extractor nothing is generated in the images directory. 
Here is the extractor log:

```
root@5374e82a938c:/FirmAFL/firmadyne#     ./sources/extractor/extractor.py -b Netgear -sql 127.0.0.1 -np -nk "WNAP320 Firmware Version 2.0.3.zip" images
>> Database Image ID: 2

/FirmAFL/firmadyne/WNAP320 Firmware Version 2.0.3.zip
>> MD5: 51eddc7046d77a752ca4b39fbda50aff
>> Tag: 2
>> Temp: /tmp/tmp3ta6ucya
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True

>>>> Zip archive data, at least v2.0 to extract, compressed size: 1197, uncompressed size: 2667, name: ReleaseNotes_WNAP320_fw_2.0.3.HTML
>> Recursing into archive ...

/tmp/tmp3ta6ucya/_WNAP320 Firmware Version 2.0.3.zip.extracted/WNAP320_V2.0.3_firmware.tar
        >> MD5: 6b66d0c845ea6f086e0424158d8e5f26
        >> Tag: 2
        >> Temp: /tmp/tmp374lzlta
        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
        >>>> POSIX tar archive (GNU), owner user name: "gz.uImage"
        >> Recursing into archive ...

/tmp/tmp374lzlta/_WNAP320_V2.0.3_firmware.tar.extracted/kernel.md5
                >> MD5: 0e15e5398024c854756d3e5f7bc78877
                >> Skipping: text/plain...

/tmp/tmp374lzlta/_WNAP320_V2.0.3_firmware.tar.extracted/root_fs.md5
                >> MD5: b43dc86ce23660652d37d97651ba1c77
                >> Skipping: text/plain...

/tmp/tmp374lzlta/_WNAP320_V2.0.3_firmware.tar.extracted/rootfs.squashfs
                >> MD5: 7ce95b252346d2486d55866a1a9782be
                >> Tag: 2
                >> Temp: /tmp/tmpds9j9nsg
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >>>> XAR archive, version: -6057, header size: 2664, TOC compressed: 18154158142782153979, TOC uncompressed: 10765983841730652167
                >> Recursing into archive ...
                >>>> Squashfs filesystem, big endian, lzma signature, version 3.1, size: 4433988 bytes, 1247 inodes, blocksize: 65536 bytes, created: 2011-06-23 10:46:19
                >>>> LZMA compressed data, properties: 0x5D, dictionary size: 65536 bytes, uncompressed size: 65536 bytes
                >> Recursing into compressed ...

/tmp/tmpds9j9nsg/_rootfs.squashfs-0.extracted/77
                        >> MD5: ba218ef71360f7d34788fd87ec596285
                        >> Skipping: application/x-executable...

/tmp/tmpds9j9nsg/_rootfs.squashfs-0.extracted/4630
                        >> MD5: ce223cfc9f0fa01defe8909901daa84b
                        >> Tag: 2
                        >> Temp: /tmp/tmpfdnecs2m
                        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                        >> Recursing into archive ...
                        >>>> Zlib compressed data, compressed
                        >> Recursing into compressed ...

/tmp/tmpfdnecs2m/_4630.extracted/1C6181
                                >> Skipping: recursion depth 4

/tmp/tmpfdnecs2m/_4630.extracted/1C9CB0
                                >> Skipping: recursion depth 4

/tmp/tmpfdnecs2m/_4630.extracted/1CE205
                                >> Skipping: recursion depth 4

/tmp/tmpfdnecs2m/_4630.extracted/1CE44B
                                >> Skipping: recursion depth 4

[... 215 lines cut ...]
        >> Tag: 2
        >> Temp: /tmp/tmpnvsdu_5p
        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
        >> Recursing into archive ...
        >> Recursing into compressed ...
        >> Cleaning up /tmp/tmpnvsdu_5p...

/tmp/tmp3ta6ucya/_WNAP320 Firmware Version 2.0.3.zip-0.extracted/116F39
        >> MD5: 34a9d95adec5606f55feaed620b84dc2
        >> Tag: 2
        >> Temp: /tmp/tmp7czpzcie
        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
        >> Recursing into archive ...
        >> Recursing into compressed ...
>> Skipping: recursion breadth 5
        >> Cleaning up /tmp/tmp7czpzcie...
>> Skipping: completed!
>> Cleaning up /tmp/tmp3ta6ucya...
root@5374e82a938c:/FirmAFL/firmadyne#
root@5374e82a938c:/FirmAFL/firmadyne# ls images/
README.md
```

I've tried with both versions of squashfs reported in the README. Could you help me?
Thanks a lot.

micsthepick, 2025-02-03:
what I've done is to manually extract with binwalk, and once I get the `img-1259692955_vol-ubi_rootfs.ubifs` , then let the tool run with that

micsthepick, 2025-02-03:
full path: _RBR350-V4.4.1.29_3.10.80_signed.img.extracted/ubifs-root/40AC8.ubi/_img-1259692955_vol-kernel.ubifs.extracted